ArduPilot/MissionPlanner #2065: SmartRTL mode disappeared

lebarsfa, 2019-01-02:
#### Issue details
Sorry if it has already been reported but I could not find a related issue : the latest Mission Planner version (1.3.58 was OK, 1.3.62 is not OK) sometimes does not seem to evoke any more the SmartRTL mode (e.g. not in Set Mode button list, FS_THR_ENABLE param description and probably elsewhere). I found also other people talking about it on [https://discuss.ardupilot.org/t/copter-3-6-2-released/35476/16](https://discuss.ardupilot.org/t/copter-3-6-2-released/35476/16).

#### Version
1.3.62

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quadrotor

#### Hardware type
HKPilot32 with ChibiOS and Copter V3.6.4 (6c4da9e3), SITL with Copter V3.7.0-dev (5f9de3cf)

#### Logs
N/A

meee1, 2019-01-09:
ok so the issue is that the parameter docs need updating on your newer install.

please open MP
press control-f
then click paramgen
wait
restart MP and it should be working again.

lebarsfa, 2019-01-09:
Many thanks, it works!