ArduPilot/MissionPlanner #2066: Set Home menu and Set Home sub-menu in FLIGHT DATA view does not always do the same as in FLIGHT PLAN

lebarsfa, 2019-01-02:
#### Issue details
Assume that you launch a SITL Copter that has SIM_GPS_DISABLE to 1 or a real drone without GPS. in Mission Planner, we can use "Set Home Here >> Set EKF Origin Here" to specify where the drone is on the Earth. We can change the Home correctly in the FLIGHT PLAN view, but in the FLIGHT DATA view, if we click on either the "Set Home Here" menu or the "Set Home Here >> Set Home Here" submenu, it changes the altitude estimation of the drone (probably a -altitude is applied to the current altitude).
This might be related to #1901.

#### Version
1.3.62

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quadrotor

#### Hardware type
HKPilot32 with ChibiOS and Copter V3.6.4 (6c4da9e3), SITL with Copter V3.7.0-dev (5f9de3cf)

#### Logs
N/A

lebarsfa, 2019-01-08:
> Also, maybe it is a misunderstanding from me but when we switch to RTL, the drone does not seem to always go to the modified home displayed in Mission Planner

This might be related to https://github.com/ArduPilot/ardupilot/issues/9498 , https://github.com/ArduPilot/ardupilot/issues/9725 .

meee1, 2019-01-09:
@rmackay9 currently i send EKF Origin with the location SRTM altitude. should this be 0?
ie if a use clicks on the map with a ground alt of 200m, i send the EKF origin as 200m.