ArduPilot/MissionPlanner #2069: Incorrect FFT frequencies

tridge, 2019-01-07:
The FFT option in MissionPlanner is producing incorrect frequencies from the ISBD and ISBH data. I've put a log here that has exact frequency noise in it:
http://uav.tridgell.net/FFT/fft-x100-y150-z200.bin
in that log the X axis (both accel and gyro) has 100Hz vibration, the Y axis has 150Hz vibration and the Z axis has 200Hz vibration. With MAVExplorer I get this:
[image]
In MissionPlanner I get quite different frequencies.
See this PR for the way to generate the exact vibration freq logs:
https://github.com/ArduPilot/ardupilot/pull/10189
Note that SITL quite deliberately sets the sample rates of the two IMUs differently, to better emulate a Pixhawk1. The rates are:
 
- accel0: 1000Hz
- accel1: 800Hz
- gyro0: 1000Hz
- gyro1: 760Hz

These rates are reflected in the smp_rate numbers in ISBH messages.
I've seen the same affect of bad frequencies in FFT in MP on real aircraft logs as well.

rmackay9, 2019-01-07:
Great to find this issue!

tridge, 2019-01-07:
I've added another log with fast sampling enabled, so we get 8x sampling gyros and 4x on accel.
http://uav.tridgell.net/FFT/fft-x100-y150-z200-fast-sample.bin

meee1, 2019-01-09:
seems to work well after the first change, and pushed a new beta

tridge, 2019-01-09:
I've done a test with this log:
http://uav.tridgell.net/FFT/fft-x100-y150-z200-fast-sample-last-50.bin
that is just the 2nd half of this log:
http://uav.tridgell.net/FFT/fft-x100-y150-z200-fast-sample.bin
with MAVExplorer I get the same Y values with both logs. With MP I get half the value on the Y axis when I use the half-sized log. That implies that MP isn't scaling for number of samples.