ArduPilot/MissionPlanner #2070: Copter Extended Tuning screen Ch7_OPT/Ch8 Opt update for Copter-3.7

rmackay9, 2019-01-11:
Similar to [this issue for Rover](https://github.com/ArduPilot/MissionPlanner/issues/2023), Copter's Extended Tuning screen should be updated to check the RC7_OPTION and RC8_OPTION parameters.  Note that this is not a simple rename because the meanings of the values have also been changed.  Thanks!

[image]

rmackay9, 2019-07-19:
This comes up fairly regularly.  It came up again today in [this posting](https://github.com/ArduPilot/ardupilot_wiki/issues/1848).

meee1, 2019-07-19:
ive added support for both the old and new name, this will still depends on the param docs used inside MP however. beta MP = master. stable MP = stable branch

rmackay9, 2019-07-19:
@meee1, great, thanks!