ArduPilot/MissionPlanner #2073: Waypoint load fail to arduplane 3.8.5

jace26, 2019-01-15:
#### Issue details
when loading wps to an arduplane w/ 3.8.5 load almost always fails on first load after establishing a connection unless a "read wps" is done first. Obviously, this clears the mission and you have to reload it before sending. 

#### Version
1.3.62    -confirmed not an issue with 1.3.56

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Sub

#### Airframe type
vtol_plane - octoquad

#### Hardware type
pixhawk 2.1

meee1, 2019-01-16:
could you upload a tlog of the failure? im interested in what could be causeing this

jace26, 2019-01-28:
@meee1 Sorry I'm just getting this: happens with SITL and real aircraft alike, here is a log from SITL:
[2019-01-28 09-04-33.txt](https://github.com/ArduPilot/MissionPlanner/files/2803591/2019-01-28.09-04-33.txt)

vascodx, 2019-02-11:
Same problem here, Arduplane 3.8.2 and Mission Planner 1.3.62

kshahar, 2019-02-21:
@meee1 This seems to be an issue with ArduCopter as well. Tested with ArduCopter SITL 3.4.4.

As a workaround, doing "Read WPs" before "Write WPs" seems to resolve this issue.

meee1, 2019-02-21:
the issue appears to be how ArduPilot handles the request

2019-01-28T23:05:36.866 | FE | 4 | 0 | 0 | 23 | FF | BE | 2C | mavlink_mission_count_t | count | 3 | target_system | 1 | target_component | 1 | mission_type | 0 |   | sig | Len | 12 |  
2019-01-28T23:05:36.928 | FE | 4 | 0 | 0 | 2E | 1 | 1 | 28 | mavlink_mission_request_t | seq | 0 | target_system | 0 | target_component | 0 | mission_type | 0 |   | sig | Len | 12 |

MP requests to target_system | 1 | target_component | 1
but the response from the autopilot is target_component | 0 | mission_type | 0
i explicitly check these to allow multiple mission uploads to different vehicles at the same time.

i need to confirm is this has always been an issue or not

magicrub, 2019-02-21:
@meee1 I've been hearing the topic of compIDs coming up lately. I have a feeling something might have changed on ArduPilot but can't point to any issues/PRs