ArduPilot/MissionPlanner #2074: Radio Calibration Bug & Improvment

SwiftGust, 2019-01-17:
#### Issue details
_Please describe the problem, or desired feature_

Radio Calibration is mandatory thing to set up airframes
and most of users does use equal or less than 8 Channel RC receivers.
However, current MP does update every 16 input channels available in ArduPilot 
regardless of channel input is updated or not,
so consequently making some annoying error in 
PreArm RC Check for RC channel 9 to 16 trim pwm (usually 974) is less than min pwm.

I think this should be updated based on RC channel available maybe checking certain channel pwm input is out of range or whether channels updated

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[ V ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Hexa
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

SwiftGust, 2019-02-13:
Looks like this problem only happens when using SBUS signals which all channels in a single stream
still, I think this kind of handling required and QGC doesn't exhibit this kinda problem

meee1, 2019-02-13:
thanks for the clarification. can I get a tlog of you ding the calibration? I just want to inspect what is sent

SwiftGust, 2019-02-13:
Hi, here I attached with RC Calibrations with 10 Hz telemetry rates

 Thank you for your effort

[1.zip](https://github.com/ArduPilot/MissionPlanner/files/2859069/1.zip)