ArduPilot/MissionPlanner #2087: Mission Planner destroys the window it was invoked from

robertschulman1, 2019-02-17:
#### Issue details
At startup, on my platform, Mission Planner typically destroys the window it was invoked from. I am running on a Win10 platform. If I start MP from a cmd window, MP starts (and brings up the logo'd MP window) and then destroys the cmd window and then brings up the MP screen users use. If I start MP from an Ubuntu window (this environment is also known as WSL, or Windows Subsystem for Linux), the terminal window Ubuntu provides *typically* dies in the same manner as the cmd window - right before the usable MP screen appears. I say "typically", because, well, sometimes, the Ubuntu terminal window does stick around and is therefore usable. If I start MP from a Cygwin window, all is OK (but the Cygwin environment lacks tools the Ubuntu environment has, which is why I prefer Ubuntu).

In any event, destroying another apps window seems wildly unfriendly and is, I believe, a bug.

#### Version
1.3.62.0

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
N/A as MP hasn't connected yet.

#### Hardware type
N/A as MP hasn't connected yet.

#### Logs
N/A

meee1, 2019-02-18:
goto MP help screen, and tick show console window

robertschulman1, 2019-02-18:
Right. Works like a champ. Thanks for the quick and effective help.