ArduPilot/MissionPlanner #2089: Question about "point camera here" mission planner function

roque-canales, 2019-02-19:
Hello,

Could you please help me to found what commande is send by mission planner to MAV when we click on "point camera here" ?

Thank you for your help.

rmackay9, 2019-02-19:
I think it is a DO_SET_ROI within a COMMAND_LONG

roque-canales, 2019-02-19:
Hello Randy,

Thank you so much.

I'm looking for the easiest way to make the drone point the camera to another drone (connected together through mavlink) automatically.

rmackay9, 2019-02-19:
@roque-canales, we've actually been thinking of adding this to [follow mode](http://ardupilot.org/copter/docs/follow-mode.html#follow-mode).  I don't know if you're happy to have the drone follow the other drone as well as point it's camera at the leader... but in any case, somewhere in the code we would need to call

> copter.camera_mount.set_roi_target(roi_location);

Where "roi_location" is a Location structure with the lat, lon and altitude of the target.

If it were added to Follow mode, I'd probably:

- add another option to the AP_Follow library's FOLL_YAW_BEHAVE parameter to allow the user to indicate they want to "point camera at Lead"
- add somewhere in mode_follow.cpp a calculation to get the lead vehicle's estimated location (either use the current vehicle's location and add the "dist_vec" value or add a new method to AP_Follow to directly return a Location structure.  Then I'd pass this into the copter.camera_mount.set_roi_target() method.

linkevinlin1, 2019-11-05:
Hi,

I would like to ask a related question. Sorry for hijacking the topic.

We would like to control the camera through a companion computer. The computer is connected to Pixhawk through USB port. Our intended way of communication is to issue "Point camera here" command from mission planner -> Pixhawk -> computer -> camera. However, the computer failed to get DO_SET_ROI from COMMAND_LONG, while I'm able to get camera trigger command from COMMAND_LONG. Is there anything special I need to set up because of the gimbal?

Thanks.