ArduPilot/MissionPlanner #2097: Geographical view is not appearing after selecting flight plan

Pujita123, 2019-03-08:
Geographical view is not appearing after selecting flight plan
We are Is_scripting and packaging team, BASF. We have a received a request for scripting mission planner 1.3.9. We have installed successfully by connecting device, but after launching the shortcut, user is not getting the geographical view under the option selection flight plan.

Version
Mission planner 1.3.9

Platform
Windows 10

[FlightPlane error screens.docx](https://github.com/ArduPilot/MissionPlanner/files/2946271/FlightPlane.error.screens.docx)

meee1, 2019-03-08:
update mission planner is the only solution to this problem

Pujita123, 2019-03-08:
Thanks for the response. We will try once and get back to you

Pujita123, 2019-03-08:
Could you provide us any email id to contact the development team of mission planner. So that we can communicate with you from our official mail id.

meee1, 2019-03-08:
please use the email address that is listed when you click the donate button.

posting email's in a public forum i try and avoid

closed 2019-03-11 by meee1