ArduPilot/MissionPlanner #2099: Dataflash viewer missing some events

rmackay9, 2019-03-15:
I think MP's display of events is missing some recent additions that are available in Copter-3.6 including

    DATA_GPS_PRIMARY_CHANGED = 67,
    DATA_WINCH_RELAXED = 68,
    DATA_WINCH_LENGTH_CONTROL = 69,
    DATA_WINCH_RATE_CONTROL = 70,
    DATA_ZIGZAG_STORE_A = 71,
    DATA_ZIGZAG_STORE_B = 72,
    DATA_LAND_REPO_ACTIVE = 73,

[the list can be found here in AP_Logger.h](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Logger/AP_Logger.h#L87) and is now shared across all vehicles.

rmackay9, 2020-03-02:
It seems like all events are missing in MP's beta firmware when used with Copter-4.0.3.
[image]

meee1, 2020-03-02:
traced this to the filtered view only. the main view is evaluating the extra data. the filtered view was not.

rmackay9, 2020-03-02:
Ah, excellent, thanks!