ArduPilot/MissionPlanner #2103: COM Port Not Appearing/Custom Firmware Issue

1iggy2, 2019-03-20:
#### Issue details
_Please describe the problem, or desired feature_
I was having difficulty finding the Install Firmware page on my old install. I was attempting to install a "Custom" version of ArduPilot. (In this case my own Ubuntu/waf build of ArduPlane). Since I couldn't go through the Install Firmware page, then the Load Custom Firmware option I decided to reinstall Mission Planner. I reinstalled three ways, first from the Mission Planner Latest Installer, then through the github link to the most recent version, currently from this page selecting "MissionPlanner-1.3.63.msi" each time I run into the same issue, When I plug in the device the COM port, in my case COM4, flashes as available and then disappears. I saw that this is commonly a driver issue and I made sure all my driver were up to date through the device manager. Now I am unable to connect to my Pixhawk4 to flash my fake custom firmware to test my ability to modify the ArduPilot source code. Please for the love of god help me.

Additional Note: The initial install of Mission Planner on my machine was from the Microsoft Store. After each iteration of an attempt to fix my issue the previous version of Mission Planner was uninstalled using the Windows Add or Remove Programs tool.

#### Version
_What version was the issue encountered with_
Holybro Pixhawk4
ASUS Windows 10 Laptop
Mission Planner 1.3.63 built 1.3.7014.32511
Arduplane, Custom compiled firmware (I kept full notes for my project team and I will provide them below)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quadplane Octo Xish
I am attempting to customize this eventually to allow for an Octo "I" configuration. See my other issue report for more information (https://github.com/ArduPilot/ardupilot/issues/10809). If you are so inclined to help with the issues I commented at the end of the post I would GREATLY appreciate it.

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Holybro Pixhawk4

#### Logs
_Please provide a link to any relevant logs that show the issue_
[image]
This is an image of my screen when I select the COM4 port when it appears and hit connect before it disappears, When COM4 appears I get the upbeat musical tone of connection and then when it disappears I get the down musical tone of disconnection.

1iggy2, 2019-03-22:
I reloaded the original firmware and the issue went away. With the "custom" firmware the issue persists

magicrub, 2019-03-22:
Well, what COM port is it? Which ones are available? The error is from trying to talk to a com Port that isn't there.

1iggy2, 2019-03-22:
So the COM Port would connect and disconnect in five seconds. During that time I'd try to connect mission planner to it. That explains the issue of trying to connect to a disconnected COM port. Now when I flash my own build of the base ardupilot github mission planner will try to connect but the MAVlink connection will timeout with no heartbeat detected.

1iggy2, 2019-03-22:
This even happens when I build it using Waf using the commands, 
./waf configure --board Pixhawk4 
./waf plane --upload
The screen displays:
[image]
Then I go into mission planner and select the correct COM port, (which now appears) and make sure the baud rate is at 115200. Then I select CONNECT. It says CONNECTING TO MAVLINK. It tries to connect for 30 seconds and fails with "No Heartbeat Packets Received"
When I select details I get this popup:
[image]
If I connect the board to QGroundControl it simply says, "Waiting for Vehicle Connection" and never finds the board.

What is the compile_commands.json? How else can I build the firmware so that it works? I will need to modify the source code for my application, and if I cannot flash the board and connect to mission planner I cannot complete my task. Is there a way for someone else to build files that will work and I can simply flash them through the "Load custom firmware" option (This was my original attempt, I would build a .apj and attempt to upload that through both Mission Planner and QGroundControl, but a .bin fil for QGroundControl,  and neither allowed for connection)

1iggy2, 2019-03-22:
@magicrub 
Sorry to tag you, but I really need help resolving this issue and I provided some more information.
Thanks!

magicrub, 2019-03-22:
So it works with stock code but not with your custom code? Well, sounds like your custom code might be involved. Could you share it?

magicrub, 2019-03-22:
This is probably not a mission planner problem. The couple second connection is the bootloader. After that, the app runs. If the app is dead, I suspect your custom code.

1iggy2, 2019-03-22:
So the "custom" code is simply me following the above instructions and I believe cloning the current ArduPilot github files. In no step did I modify the contents of those files. I'll include the arduplane files I have and the bin in my Pixhawk4 folder that was under build.

[ArduPlane.zip](https://github.com/ArduPilot/MissionPlanner/files/2997873/ArduPlane.zip)
[bin.zip](https://github.com/ArduPilot/MissionPlanner/files/2997874/bin.zip)

magicrub, 2019-03-22:
This is not a bug in Mission Planner or ArduPilot. I suspect it's just a USB driver you're missing from ChibiOS. It's included in Mission Planner, perhaps a reinstall would solve it.

If that doesn't fix it, please post on discuss.ardupilot.org and mention important things like:
- how you built the code (the commands you used),
- computer OS
- version of code (git hash)

closed 2019-03-22 by meee1

1iggy2, 2019-03-22:
Hey @magicrub I've uninstalled and reinstalled Mission Planner approx 12 times, I've checked all my drivers for updates and even deleted some that came with Mission Planner to force a reinstall. I've handed off the Pixhawk4 to a teammate so he can work on it on his system. Where can I find the git hash? Would it be sufficient to link to this post to provide my information on the forum as this is a decently complete document of my attempts, including an entire document how I set up my build environment and all the commands I used, also I've provided all my commands I've used and my troubleshooting attempts.

magicrub, 2019-03-22:
Please post help questions on our forum at discuss.ardupilot.org. Github issues are for confirmed bugs and feature requests. This is a basic USB driver problem on your computer. Please post on forums