ArduPilot/MissionPlanner #2126: MP 1.3.64 won't start on linux

davidbuzz, 2019-04-06:
#### Issue details
mono MissionPlanner.exe won't start on linux. 

#### Version
1.3.64,  I also tried 1.3.62, same thing.

#### Logs
MONO_LOG_LEVEL=debug mono MissionPlanner.exe > mono_mp.log
  gives this when I start MP, the splash screen opens, I count to about 10 seconds, and then press ctrl-c on the terminal that started it:
see attached log.
[mono_mp.log](https://github.com/ArduPilot/MissionPlanner/files/3050371/mono_mp.log)

davidbuzz, 2019-04-06:
I just tried 1.3.58 and it starts fine.

davidbuzz, 2019-04-06:
after successfully starting 1.3.58, now 1.3.64 now starts properly too.

meee1, 2019-04-06:
just pushed a change that "might" fix the issue. but its a guess

kikislater, 2019-04-07:
1.3.62 works on archlinux

davidbuzz, 2019-04-08:
today I have tried all three I have downloaded, 1.3.58,1.3.62, and 1.3.64, and all three fail to proceed past the splash screen.  I tried doing a "rm -rf ~/Mission\ Planner/" and the following ( at least 3 )  startup attempts worked.

davidbuzz, 2019-04-08:
The inability to start bit me again, so I tried deleting all the subfolder/s from inside ~/Mission\ Planner ( cd ~/Mission\ Planner ; rm -rf gmapcache logs sitl srtm ) , instead of the entire thing, and MP started fine again after that... so it's inside one of these folders the problem exists.

meee1, 2019-04-08:
what mono version?

davidbuzz, 2019-04-08:
mono --version
Mono JIT compiler version 4.6.2 (Debian 4.6.2.7+dfsg-1ubuntu1)
... which is the latest one that is packaged and generally available for Ubuntu 18.04.2 LTS

davidbuzz, 2019-04-08:
SInce I had luck deleting randome folders in ~/Mission\ Planner/ , I thought I'd try deleting one folder at a time till it didn't hang on boot... and I started with deleting the  'logs' folder, and that seems to have fixed it, at least once so far:
cd ~/Mission\ Planner/
 rm -rf logs

meee1, 2019-04-08:
do a lsof to see what file its reading at the time

davidbuzz, 2019-04-09:
I ran 1.3.64 this morning, and it *again* hung on startup.
attached is the output from 'lsof' with what appeared to be spurious parts removed.  I ran this once while the splash screen was "stuck", and again after terminating it, removing the ~/Mission\ Planner/logs/ folder and restarting it.
sudo lsof +c 0 -w | grep mono | grep -v "/usr/share/icons" | grep -v "/usr/lib/mono"

[1_lsof_while_hung_on_startup.txt](https://github.com/ArduPilot/MissionPlanner/files/3056706/1_lsof_while_hung_on_startup.txt)

[2_lsof_after_good_startup.txt](https://github.com/ArduPilot/MissionPlanner/files/3056705/2_lsof_after_good_startup.txt)

davidbuzz, 2019-04-10:
rm -rf ~/Mission\ Planner/logs/*log  before each startup prevents  my issue from occuring.

meee1, 2019-04-10:
if you have narrowed it to the log directory, then can you provide a sample log thats causing it?

meee1, 2019-04-11:
also have you checked the latest beta?