ArduPilot/MissionPlanner #2127: nmea out in MP not work in Reefmaster for live mapping

plastic11, 2019-04-08:
#### Issue details
Nmea out in MP is not work in Reefmaster for live mapping (depth mapping soft). Meanwhile DrDepth (another depth mapping software) works only if ignore nmea sum is checked. Perhaps the “Ingore NMEA checksum” is a clue - perhaps MP’s NMEA output isn’t providing a correct checksum.

#### Version
All

#### Platform
[ x ] All

#### Airframe type
All

#### Hardware type
All

#### Logs
no logs

plastic11, 2019-04-29:
Thank you. Now everything works just fine now.

plastic11, 2020-04-02:
For now nmea out is not readable in other software.

that2, 2021-06-29:
> Thank you. Now everything works just fine now.

Hi plastic11, having the same issue right now, what did you make in order to solve it? thanks!