ArduPilot/MissionPlanner #2141: opt_* messages doesn't update in Status tab

bys1123, 2019-05-04:
#### Issue details
I'm not very sure this is cause by MP or Ardupilot. 

MP can only show the first frame from optical flow. But the log shows it works well.

This make lot of people think their flow sensor is dead. 

#### Version
Copter-3.6.7~3.6.9

#### Platform
[  ] All
[  ] AntennaTracker
[√] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Hardware type
Pix32(Pixhawk), Pixhawk4

[image]

meee1, 2019-05-04:
do you have a tlog that you think has the issue?

bys1123, 2019-05-05:
@meee1 Weird, my mission planner didn't storage tlog too, I can see 0kb tlog and rlog in my document folder while I connect the vihicle, but when I disconnect, it's gone.

bys1123, 2019-05-05:
https://www.dropbox.com/s/8mos4rk87wtbm38/2019-05-04%2021-15-39.tlog?dl=0
Here is my tlog, Thanks!

bys1123, 2019-05-05:
https://discuss.ardupilot.org/t/how-to-setup-and-check-optical-flow/20444/40
Here is other person have the same issue.

meee1, 2019-05-06:
the log you provided only has 5 samples of data in it

2019-05-04T21:15:41.359 | FE | 1A | 0 | 0 | 55 | 1 | 1 | 64 | mavlink_optical_flow_t | time_usec | 7881 | flow_comp_m_x | -0.01688853
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2019-05-04T21:15:41.361 | FE | 1A | 0 | 0 | 61 | 1 | 1 | 64 | mavlink_optical_flow_t | time_usec | 7896 | flow_comp_m_x | -0.01688853
2019-05-04T21:15:41.363 | FE | 1A | 0 | 0 | 6D | 1 | 1 | 64 | mavlink_optical_flow_t | time_usec | 7901 | flow_comp_m_x | -0.01688853
2019-05-04T21:15:41.363 | FE | 1A | 0 | 0 | 79 | 1 | 1 | 64 | mavlink_optical_flow_t | time_usec | 7919 | flow_comp_m_x | -0.01688853
2019-05-04T21:15:41.364 | FE | 1A | 0 | 0 | 86 | 1 | 1 | 64 | mavlink_optical_flow_t | time_usec | 7920 | flow_comp_m_x | -0.01688853

bys1123, 2019-05-06:
@meee1 Thanks, but I don't know why it's stoped, but in SD log is fine.

bys1123, 2019-05-06:
I think maybe I should find Ardupilot team.

bys1123, 2019-05-12:
[image]
[image]
Tried CX-OF today, values of opt_* still not updating.