ArduPilot/MissionPlanner #2143: mono MissionPlanner.exe doesn't show .BIN logs when DataFlash log review

Jaaaky, 2019-05-11:
#### Issue details
_On Linux `mono MissionPlanner.exe` doesn't show .BIN logs when DataFlash log review
The file browser doesn't show any .BIN extensions by default and I've to type *.BIN in each directory to view any DF log files_

#### Version
_Mission Planner 1.3.66_
_Mono 5.20_

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2019-05-11:
will be a case issue.
what case is the file on the filesystem?

Jaaaky, 2019-05-11:
On Linux everything is case sensitive by default. Even FAT32 filesystem.

Jaaaky, 2019-05-11:
And DF logs extension is always upper case .BIN

meee1, 2019-05-11:
ok, so always uppercase

Jaaaky, 2019-05-11:
I can confirm it's a case issue.
I could work-around it by `export MONO_IOMAP=drive:case` .. found through `man mono`

meee1, 2019-05-11:
cool thanks for that feedback
ill add a note on MP startup

closed 2019-06-14 by Jaaaky