ArduPilot/MissionPlanner #2148: MP 1.3.6x bad graphing of BIN logs if LOG_DISARMED=1

Jaaaky, 2019-05-18:
MissionPlanner versions 1.3.6x doesn't graph properly specially if LOG_DISARMED is On.

It just displays single vertical point.

Attached are two screenshots from the graphing of BARO.temp - just an example - for the same BIN log file using MP 1.3.57 and 1.3.66

[image]
[image]

Jaaaky, 2019-05-18:
After posting I found that when I un-check "Use Time" it works!!
It's still a bug.

meee1, 2019-05-18:
I don't think it's a bug at all.  The log entry's don't have a valid time