ArduPilot/MissionPlanner #2149: Support for NXP FMUK66 USB VID/PID

jarivanewijk, 2019-05-20:
NXP Semiconductors has made the RDDRONE-FMUK66 for its HoverGames competition. The board is already supported by PX4 and hopefully will be supported in Ardupilot in the future (I believe there was some early effort some time ago). 

It would be great if support for the following USB VID/PID could be added to the Mission Planner drivers.

USB_VENDOR 0x1FC9
USB_PRODUCT 0x1C
USB_STRING_MANUFACTURER "NXP"
USB_STRING_PRODUCT "RDDRONE-FMUK66"

Thanks. Please let us know if we can contribute in any way, or if more information is needed.

FYI @igalloway, @LeoMustafaNXP, @davids5, @DonLakeFlyer

jarivanewijk, 2019-06-05:
I found some time to look into this myself. I made a pull request (#2155). Please verify that I made the right changes. Thanks!

meee1, 2019-06-05:
is the board using the ardupilot bootloader?

davids5, 2019-06-05:
@meee1 It is using the PX4 bootloader built on top of the NXP_Kinetis_Bootloader_2_0_0.

jarivanewijk, 2019-06-07:
@meee1 It shouldn't matter whether the board is using the ArduPilot or PX4 bootloader, right?

What we want right now is to get FMUK66 supported in the driver, and to have the board being recognized by Mission Planner. Firmware update through Mission Planner would be nice, but is not really a priority right now.

Could you verify that I made the right changes to achieve this in #2155 or explain what else needs to be done?