ArduPilot/MissionPlanner #2153: Bug: Battery Voltage always displayed as 0.00V

cheintz, 2019-06-04:
#### Issue details
MissionPlanner displays BAT 0.00v (but reads current correctly). This is in the "status" tab, as well as user items.

Connecting with QGroundControl to the same aircraft shows the correct voltage.

#### Version
1.3.66 build 1.3.7054.36589
Windows 7 x64

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Flying Wing

#### Hardware type
PX4

#### Logs
n\a

meee1, 2019-06-04:
could you try beta MP?

cheintz, 2019-06-11:
Uninstall and reinstall fixed it. Previously what I had was upgraded, but I'm not sure what the original version was.

rmackay9, 2019-06-11:
Great, thanks for the follow up, closing..

closed 2019-06-11 by rmackay9