ArduPilot/MissionPlanner #2156: Boring BAD AHRS and GPS Healt during SITL

robustini, 2019-06-06:
#### Issue details
During built-in SITL with X-Plane these AHRS errors continue to appear, this obviously clears any other message and is really boring.
AHRS_EFK_TYPE cheating already set to 10, the problem is there anyway.

https://youtu.be/zOJVnTfiBr8

#### Version

The latest beta

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Glider

#### Hardware type
SITL
#### Logs
Log is related this simulated flight

[00000004.ZIP](https://github.com/ArduPilot/MissionPlanner/files/3260525/00000004.ZIP)

magicrub, 2019-06-06:
bad GPS will lead to bad AHRS. The question becomes: why is the GPS bad in the sim?

magicrub, 2019-06-06:
params look OK, I think those are all defaults

```
GPS_AUTO_CONFIG	1
GPS_AUTO_SWITCH	1
GPS_BLEND_MASK	5
GPS_BLEND_TC	10
GPS_DELAY_MS	0
GPS_DELAY_MS2	0
GPS_GNSS_MODE	0
GPS_GNSS_MODE2	0
GPS_INJECT_TO	127
GPS_MIN_DGPS	100
GPS_MIN_ELEV	-100
GPS_NAVFILTER	8
GPS_POS1_X	0
GPS_POS1_Y	0
GPS_POS1_Z	0
GPS_POS2_X	0
GPS_POS2_Y	0
GPS_POS2_Z	0
GPS_RATE_MS	200
GPS_RATE_MS2	200
GPS_RAW_DATA	0
GPS_SAVE_CFG	0
GPS_SBAS_MODE	2
GPS_SBP_LOGMASK	-32768
GPS_TYPE	1
GPS_TYPE2	0

```

meee1, 2019-06-12:
the message is displayed based on what MP is told
!sensors_health.ahrs && sensors_enabled.ahrs && sensors_present.ahrs

the "sensor" is enabled, present and not health.
so there are 2 solutions. either make xplanes sitl not send this status to MP. or add a exclusion for sitl for MP to ignore.

my preferred approach would be not get bad data in the first place, and resolve the underlying issue. vs just sweeping it under the rug.

robustini, 2019-06-12:
Thanks Michael for reply.
So it's a problem that @tridge  should solve?