ArduPilot/MissionPlanner #2160: MP freezes on some windows

Jaaaky, 2019-06-11:
#### Issue details
MissionPlanner always freezes on pop-up status variable windows on flight data.
When you need to change some variable on "quick" tab, tuning or User Item on HUD.
The pop-up window loads very slowly and MP is completely frozen then.
It's very critical when some change is needed during flight.

#### Version
It's an old issue, but still applies to current version 1.3.66.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2019-06-11:
fixed a few days ago
https://github.com/ArduPilot/MissionPlanner/commit/c67c69be4c35ed3d42c2328c548a14769df24218

closed 2019-06-11 by meee1

Jaaaky, 2019-06-11:
Any built available for test?

meee1, 2019-06-11:
should be part of the current beta

Jaaaky, 2019-06-11:
Beta is almost 23 days ago.
http://firmware.ardupilot.org/Tools/MissionPlanner/beta/
So I think it doesn't include the fix

meee1, 2019-06-11:
more like 1-2 years old
you need to use the built in updater

help > update beta

Jaaaky, 2019-06-11:
Is MissionPlanner official build are debug or release?
I noticed many debugging there and over whole performance is not good. Can we gain better performance with stripped release builds? And keep debugging build there for those who need to debug some issues?

meee1, 2019-06-11:
on windows you can use ngen
on linux/mono you can use mono --aot

ahead of time compiling.

Jaaaky, 2019-06-11:
On Windows; The following is correct Or something else is needed;
I've run the command two times. 2nd time displayed this:

```
PS C:\Windows\system32> C:\Windows\Microsoft.NET\Framework64\v4.0.30319\ngen.exe install 'C:\Program Files (x86)\Mission Planner\MissionPlanner.exe'
Microsoft (R) CLR Native Image Generator - Version 4.7.3190.0
Copyright (c) Microsoft Corporation.  All rights reserved.
Failed to load dependency OpenTK.GLControl of assembly MissionPlanner, Version=1.3.7054.36584, Culture=neutral, PublicKeyToken=null because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency Mono.Android of assembly UsbSerialForAndroid, Version=0.2.0.0, Culture=neutral, PublicKeyToken=null because of the following error : The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
Failed to load dependency Java.Interop of assembly UsbSerialForAndroid, Version=0.2.0.0, Culture=neutral, PublicKeyToken=null because of the following error : The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
Failed to load dependency OpenTK.GLControl of assembly MissionPlanner.Controls, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency OpenTK.GLControl of assembly SkiaSharp.Views.Desktop, Version=1.68.0.0, Culture=neutral, PublicKeyToken=0738eb9f132ed756 because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency System.Web.Mvc of assembly DotNetOpenAuth, Version=4.3.0.0, Culture=neutral, PublicKeyToken=2780ccd10d57b246 because of the following error : The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
All compilation targets are up to date.
```

meee1, 2019-06-11:
are you on windows or wine or mono?

Jaaaky, 2019-06-11:
All of them, but the command on previous post is on windows 10.
On wine 4.1 I got almost the same result.

Jaaaky, 2019-06-11:
Here is wine ngen output: For 2nd time run
```
wine ~/.wine/drive_c/windows/Microsoft.NET/Framework/v4.0.30319/ngen.exe install ./MissionPlanner.exe 
Microsoft (R) CLR Native Image Generator - Version 4.6.1590.0
Copyright (c) Microsoft Corporation.  All rights reserved.
Failed to load dependency OpenTK.GLControl of assembly MissionPlanner, Version=1.3.7101.7735, Culture=neutral, PublicKeyToken=null because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency Java.Interop of assembly UsbSerialForAndroid, Version=0.2.0.0, Culture=neutral, PublicKeyToken=null because of the following error : File not found. (Exception from HRESULT: 0x80070002)
Failed to load dependency Java.Interop of assembly Mono.Android, Version=0.0.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065 because of the following error : File not found. (Exception from HRESULT: 0x80070002)
Failed to load dependency System.Xml of assembly Mono.Android, Version=0.0.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065 because of the following error : File not found. (Exception from HRESULT: 0x80070002)
Failed to load dependency System.Runtime.Serialization of assembly Mono.Android, Version=0.0.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065 because of the following error : File not found. (Exception from HRESULT: 0x80070002)
Failed to load dependency OpenTK.GLControl of assembly MissionPlanner.Controls, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency OpenTK.GLControl of assembly SkiaSharp.Views.Desktop, Version=1.68.0.0, Culture=neutral, PublicKeyToken=0738eb9f132ed756 because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency OpenTK.GLControl of assembly Xamarin.Forms.Platform.WinForms, Version=3.4.0.0, Culture=neutral, PublicKeyToken=null because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency System.Web.Mvc of assembly DotNetOpenAuth, Version=4.3.0.0, Culture=neutral, PublicKeyToken=2780ccd10d57b246 because of the following error : File not found. (Exception from HRESULT: 0x80070002)
All compilation targets are up to date.
```

Jaaaky, 2019-06-11:
mono doesn't show errors however:

```
mono --aot -O=all MissionPlanner.exe
Mono Ahead of Time compiler - compiling assembly /home/user/Downloads/MissionPlanner-1.3.67Beta/MissionPlanner.exe
AOTID 30971E7F-611A-470B-AD4D-CDE80083A5AB
Code: 3763489(81%) Info: 109528(2%) Ex Info: 330489(7%) Unwind Info: 9401(0%) Class Info: 164556(3%) PLT: 53550(1%) GOT Info: 121804(2%) Offsets: 37206(0%) GOT: 161152
Compiled: 4308/4308 (100%), No GOT slots: 0 (0%), Direct calls: 0 (0%)
Executing the native assembler: "as" --64  -o /tmp/mono_aot_g5rITt.o /tmp/mono_aot_g5rITt
Executing the native linker: "ld"  -shared -o /home/user/Downloads/MissionPlanner-1.3.67Beta/MissionPlanner.exe.so.tmp  /tmp/mono_aot_g5rITt.o 
JIT time: 1564 ms, Generation time: 303 ms, Assembly+Link time: 1309 ms.
```

Jaaaky, 2019-06-11:
And again here is Windows 10 output for latest Beta;
```
 C:\Windows\Microsoft.NET\Framework64\v4.0.30319\ngen.exe install .\MissionPlanner.exe
Microsoft (R) CLR Native Image Generator - Version 4.7.3190.0
Copyright (c) Microsoft Corporation.  All rights reserved.
Failed to load dependency OpenTK.GLControl of assembly MissionPlanner, Version=1.3.7101.7735, Culture=neutral, PublicKeyToken=null because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency Java.Interop of assembly UsbSerialForAndroid, Version=0.2.0.0, Culture=neutral, PublicKeyToken=null because of the following error : The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
Failed to load dependency Java.Interop of assembly Mono.Android, Version=0.0.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065 because of the following error : The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
Failed to load dependency System.Xml of assembly Mono.Android, Version=0.0.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065 because of the following error : The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
Failed to load dependency System.Runtime.Serialization of assembly Mono.Android, Version=0.0.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065 because of the following error : The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
Failed to load dependency OpenTK.GLControl of assembly MissionPlanner.Controls, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency OpenTK.GLControl of assembly SkiaSharp.Views.Desktop, Version=1.68.0.0, Culture=neutral, PublicKeyToken=0738eb9f132ed756 because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency OpenTK.GLControl of assembly Xamarin.Forms.Platform.WinForms, Version=3.4.0.0, Culture=neutral, PublicKeyToken=null because of the following error : Strong name validation failed. (Exception from HRESULT: 0x8013141A)
Failed to load dependency System.Web.Mvc of assembly DotNetOpenAuth, Version=4.3.0.0, Culture=neutral, PublicKeyToken=2780ccd10d57b246 because of the following error : The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
All compilation targets are up to date.
```