ArduPilot/MissionPlanner #2161: DataFlash Log graphs incompatible with EKF3

Jaaaky, 2019-06-11:
#### Issue details
On vehicle here EKF2 is disabled and only EKF3 is enabled and used; graphing DataFlash logs with graph templates for most of EKF section is failed.
Example, EKF/Wind velocity NE

I've found that it's heavily dependent on EKF2 on mavgraphs.xml file. So `sed -i 's/NKF/XKF/g' mavgraphs.xml`  solved the problem for such logs. But now it doesn't work for EKF2 logs. So I've to manually switch xml files.

#### Version
Old problem but applicable to latest version 1.3.66 and Beta.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

Can be easily simulated on SITL, Just set EK2_ENABLE,0 and EK3_ENABLE,1 and AHRS_EKF to 3 then reboot and test.

Jaaaky, 2019-06-14:
@meee1 I can see you fixed this too? Can close?

meee1, 2019-06-14:
check latest beta. should be done

Jaaaky, 2019-06-14:
Yes, I see it working finally. Thanks a lot :)
Why do not you publish the latest beta download link? It makes it easier for beta testers.

The obsolete link [here](http://firmware.ardupilot.org/Tools/MissionPlanner/beta/) can be set for redirection to the new link on https://ci.appveyor.com

closed 2019-06-14 by Jaaaky

meee1, 2019-06-14:
the beta inside MP pulls from the CI server.

i only leave the beta folder there for legacy reasons.

i could remove it now