ArduPilot/MissionPlanner #2162: wrap_360 graph function doesn't work at all

Jaaaky, 2019-06-11:
#### Issue details
On graphing DataFlash logs, any graph that is defined as wrap_360() on mavgraphs.xml doesn't work.
It's simply skipped at graphing and not displayed.

#### Version
Tested 1.3.57, 1.3.66 and Beta
I remember it was working long time ago.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

Jaaaky, 2019-06-11:
I've tested multiple random versions down to 1.3.33 where mavgraphs.xml ever appeared. I think wrap_360() never worked.

meee1, 2019-06-12:
i think the current beta should now work with this.

NOTE beta. not 1.3.67

Jaaaky, 2019-06-14:
Thanks :+1:  It is working :)

Jaaaky, 2019-06-27:
Again wind graphs with wrap_360 doesn't work on latest beta?? It was working 13 days ago!!

Why there are always regressions in MP new releases? Every new release introduce regression bugs.