ArduPilot/MissionPlanner #2163: not able to open mp after update

gnkarn, 2019-06-11:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_
1.3.66

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

9-06-11 12:11:47,446  INFO Program - ******************* Logging Configured ******************* (:0) [1]
2019-06-11 12:11:47,675 DEBUG Program - Loaded: MissionPlanner.resources, Version=1.3.7054.36584, Culture=es-ES, PublicKeyToken=null (:0) [1]
2019-06-11 12:11:48,066  INFO Program - SkiaLoaded (:0) [1]
2019-06-11 12:11:48,295 DEBUG Program - Loaded: Accessibility, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a (:0) [1]
2019-06-11 12:11:49,899 DEBUG Program - Loaded: GMap.NET.WindowsForms, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [1]
2019-06-11 12:11:49,965 DEBUG Program - TypeResolve Failed: Mono.Runtime from MissionPlanner.Utilities, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [1]
2019-06-11 12:11:50,365  INFO Program - 64bit os False, 64bit process False (:0) [1]
2019-06-11 12:11:50,501 DEBUG Program - Loaded: System.Reactive, Version=4.0.0.0, Culture=neutral, PublicKeyToken=94bc3704cddfc263 (:0) [Base Thread]
2019-06-11 12:11:50,749 DEBUG Program - Loaded: GMap.NET.Drawing, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:50,879 DEBUG Program - Loaded: SvgNet, Version=1.0.0.0, Culture=neutral, PublicKeyToken=38326cb7e06851fc (:0) [Base Thread]
2019-06-11 12:11:51,042 DEBUG Program - Loaded: ProjNET, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:51,139 DEBUG Program - Loaded: GeoAPI.CoordinateSystems, Version=1.7.5.0, Culture=neutral, PublicKeyToken=a1a0da7def465678 (:0) [Base Thread]
2019-06-11 12:11:51,159 DEBUG Program - TypeResolve Failed: Mono.Runtime from MissionPlanner.ArduPilot, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:51,222 DEBUG Program - Loaded: System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35 (:0) [Base Thread]
2019-06-11 12:11:51,314 DEBUG Program - Loaded: Transitions, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:51,432 DEBUG Program - Loaded: OpenTK.GLControl, Version=3.0.1.0, Culture=neutral, PublicKeyToken=bad199fe84eb3df4 (:0) [Base Thread]
2019-06-11 12:11:51,786 DEBUG Program - Loaded: OpenTK, Version=3.0.1.0, Culture=neutral, PublicKeyToken=bad199fe84eb3df4 (:0) [Base Thread]
2019-06-11 12:11:51,841  INFO MissionPlanner.MainV2 - Mainv2 ctor (:0) [Base Thread]
2019-06-11 12:11:51,847  INFO MissionPlanner.MainV2 - Loading config (:0) [Base Thread]
2019-06-11 12:11:51,982 DEBUG Program - Loaded: MissionPlanner.Strings, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:52,147 DEBUG Program - Loaded: SkiaSharp.Views.Desktop, Version=1.68.0.0, Culture=neutral, PublicKeyToken=0738eb9f132ed756 (:0) [Base Thread]
2019-06-11 12:11:52,792 DEBUG Program - Loaded: SkiaSharp, Version=1.68.0.0, Culture=neutral, PublicKeyToken=0738eb9f132ed756 (:0) [Base Thread]
2019-06-11 12:11:53,248 DEBUG Program - Loaded: BSE.Windows.Forms, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:53,422 DEBUG Program - Loaded: ZedGraph, Version=5.1.5.27728, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:53,694 DEBUG Program - Loaded: ObjectListView, Version=2.5.1.30870, Culture=neutral, PublicKeyToken=b1c5bf581481bcd4 (:0) [Base Thread]
2019-06-11 12:11:53,778 DEBUG Program - Loaded: Microsoft.Win32.Registry, Version=4.1.1.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a (:0) [Base Thread]
2019-06-11 12:11:53,812 DEBUG Program - TypeResolve Failed: Mono.Runtime from MissionPlanner.Utilities, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:53,816  INFO MissionPlanner.Utilities.srtm - .cctor (:0) [Base Thread]
2019-06-11 12:11:54,064  INFO MissionPlanner.Utilities.srtm - C:\ProgramData\Mission Planner\srtm (:0) [Base Thread]
2019-06-11 12:11:54,065 DEBUG Program - TypeResolve Failed: Mono.Runtime from MissionPlanner, Version=1.3.7054.36584, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:54,066  INFO MissionPlanner.Utilities.srtm - requestRunner start (:0) [SRTM request runner]
2019-06-11 12:11:54,070 DEBUG Program - TypeResolve Failed: Mono.Runtime from MissionPlanner, Version=1.3.7054.36584, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:54,071  INFO MissionPlanner.Utilities.Speech - TTS: init, mono = False (:0) [Base Thread]
2019-06-11 12:11:54,204  INFO MissionPlanner.MainV2 - Create FD (:0) [Base Thread]
2019-06-11 12:11:54,258  INFO MissionPlanner.GCSViews.FlightData - Ctor Start (:0) [Base Thread]
2019-06-11 12:11:54,499 DEBUG Program - Loaded: System.Drawing.Common, Version=4.0.0.1, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51 (:0) [Base Thread]
2019-06-11 12:11:54,549 DEBUG Program - TypeResolve Failed: Mono.Runtime from OpenTK, Version=3.0.1.0, Culture=neutral, PublicKeyToken=bad199fe84eb3df4 (:0) [Base Thread]
2019-06-11 12:11:54,679 DEBUG Program - Loaded: mscorlib.resources, Version=4.0.0.0, Culture=es, PublicKeyToken=b77a5c561934e089 (:0) [Base Thread]
2019-06-11 12:11:54,691 DEBUG Program - FirstChanceException in: OpenTK (:0) [Base Thread]
System.DllNotFoundException: No se puede cargar el archivo DLL 'libEGL.dll': No se puede encontrar el módulo especificado. (Excepción de HRESULT: 0x8007007E)
   en OpenTK.Platform.Egl.Egl.GetCurrentContext()
   en OpenTK.Platform.Egl.Egl.get_IsSupported()
2019-06-11 12:11:54,963 DEBUG Program - Loaded: Microsoft.GeneratedCode, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:55,693 DEBUG Program - TypeResolve Failed: Mono.Runtime from GMap.NET.Core, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:55,879 DEBUG Program - Loaded: GeoUtility, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:11:56,567 DEBUG Program - FirstChanceException in: mscorlib (:0) [Base Thread]
[... 103 lines cut ...]
2019-06-11 12:12:05,923  INFO MissionPlanner.Plugin.PluginLoader - LoadFromSameFolder MissionPlanner.Gridv2, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null-> MissionPlanner.Gridv2.resources, Version=1.0.0.0, Culture=es, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:12:05,998  INFO MissionPlanner.Plugin.PluginLoader - LoadFromSameFolder MissionPlanner.Gridv2, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null-> MissionPlanner.Gridv2.resources, Version=1.0.0.0, Culture=es, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:12:06,075  INFO MissionPlanner.Plugin.PluginLoader - LoadFromSameFolder MissionPlanner.Gridv2, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null-> MissionPlanner.Gridv2.resources, Version=1.0.0.0, Culture=es, PublicKeyToken=null (:0) [Base Thread]
2019-06-11 12:12:06,117  INFO MissionPlanner.MainV2 - Load Pluggins... Done (:0) [Base Thread]
2019-06-11 12:12:06,120  INFO MissionPlanner.MainV2 - show FlightData (:0) [Base Thread]
2019-06-11 12:12:06,147  INFO MissionPlanner.GCSViews.FlightData - Activate Called (:0) [Base Thread]
2019-06-11 12:12:06,151 DEBUG MissionPlanner.Utilities.Tracking - v=1&tid=UA-43098846-1&cid=1b74528f-3218-41da-bb9a-a24237f3246a&t=appview&an=Mission+Planner&av=1.3.66&cd=MissionPlanner.GCSViews.FlightData&dp=MissionPlanner.GCSViews.FlightData&dt=FlightData&sc=start&ul=en-US&sd=32-bits&sr=1366x768&cd1=Microsoft+Windows+NT+10.0.14393.0&cd4=4&z=1451435305 (:0) [8]
2019-06-11 12:12:06,738  INFO MissionPlanner.Controls.HUD - OnLoad Start (C:\Users\mich1\Desktop\Hex\MissionPlanner\ExtLibs\Controls\HUD.cs:878) [Base Thread]
2019-06-11 12:12:06,748  INFO MissionPlanner.Controls.HUD - Vendor: Intel Corporation (C:\Users\mich1\Desktop\Hex\MissionPlanner\ExtLibs\Controls\HUD.cs:887) [Base Thread]
2019-06-11 12:12:06,750  INFO MissionPlanner.Controls.HUD - Version: 2.1 (C:\Users\mich1\Desktop\Hex\MissionPlanner\ExtLibs\Controls\HUD.cs:888) [Base Thread]
[... cut ...]
2019-06-11 12:12:06,782 DEBUG MissionPlanner.Controls.HUD - Hint (C:\Users\mich1\Desktop\Hex\MissionPlanner\ExtLibs\Controls\HUD.cs:928) [Base Thread]
2019-06-11 12:12:06,784 DEBUG MissionPlanner.Controls.HUD - Enable (C:\Users\mich1\Desktop\Hex\MissionPlanner\ExtLibs\Controls\HUD.cs:944) [Base Thread]
2019-06-11 12:12:06,786  INFO MissionPlanner.Controls.HUD - OnLoad Done (C:\Users\mich1\Desktop\Hex\MissionPlanner\ExtLibs\Controls\HUD.cs:956) [Base Thread]
2019-06-11 12:12:06,902 DEBUG MissionPlanner.Utilities.Tracking - OK (:0) [8]

meee1, 2019-06-12:
my guess is a driver issue on your tablet
most likely graphics card driver.
also please post support issues on the ardupilot forum. not here on the issue list.

closed 2019-06-12 by meee1

gnkarn, 2019-06-12:
Ok, thank you