ArduPilot/MissionPlanner #2170: Toggle Safety Switch doesn't work recently

Jaaaky, 2019-06-22:
#### Issue details
"Toggle Safety Switch" button on Testing Screen doesn't work for recent versions of MP.
It works fine on mavproxy `arm safetyoff`
#### Version
1.3.66 - 1.3.67

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Sub

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk

meee1, 2019-06-24:
looks of to me

was the MAV armed but with saftey on?
i think the flow should be disable saftey button while  not ARMED.
the arm, and it will work

Jaaaky, 2019-06-27:
It doesn't work, not armed.
mavproxy works on exact same case.
Why mavproxy works and MP not?