ArduPilot/MissionPlanner #2184: Guided_enable command in Flight plan for Arduplane

StephBMG, 2019-07-18:
#### Issue details
GUIDED_ENABLE is available in MP for arducopter flight plan but not in arduplane

#### Version
1.3.68

#### Platform
 Arduplane 3.9.8

#### Airframe type
Quadplane
#### Hardware type
Pixhawk cube

davidalois, 2019-08-03:
I also have this issue

Francis-Internest, 2019-08-06:
The Guided_Enable command is essential for the precision landing systems (LoLas) that we develop
Guided_Enable is available in Flight plan for arducopter why not for arduplane?
Thank you !

hadrien-int, 2019-08-06:
+1 The Guided_Enable command is essential for the precision landing systems (LoLas) that we develop
Guided_Enable is available in Flight plan for arducopter why not for arduplane?
Thank you !