ArduPilot/MissionPlanner #2186: bulid error in missionplanner

ashish-yadav1, 2019-07-22:
#### Issue details
This version of Visual Studio is unable to open the following projects. The project types may not be installed or this version of Visual Studio may not support them. 
For more information on enabling these project types or otherwise migrating your assets, please see the details in the "Migration Report" displayed after clicking OK.
	 - Xamarin.iOS, "E:\Ashish2\MissionPlanner\ExtLibs\Xamarin\Xamarin.iOS\Xamarin.iOS.csproj"
	 - Xamarin.Android, "E:\Ashish2\MissionPlanner\ExtLibs\Xamarin\Xamarin.Android\Xamarin.Android.csproj"
	 - UsbSerialForAndroid, "E:\Ashish2\MissionPlanner\ExtLibs\UsbSerialForAndroid\UsbSerialForAndroid.csproj"

issue 2
xtLibs\UsbSerialForAndroid\UsbSerialForAndroid.csproj: The application which this project type is based on was not found. Please try this link for further information: http://go.microsoft.com/fwlink/?LinkID=299083&projecttype=10368E6C-D01B-4462-8E8B-01FC667A7035
issue 3
Severity	Code	Description	Project	File	Line	Suppression State
Error		NuGet Package restore failed for project ExtLibs\AltitudeAngelWings: Unable to find version '2.8.0' of package 'Flurl'.
  C:\Program Files (x86)\Microsoft SDKs\NuGetPackages\: Package 'Flurl.2.8.0' is not found on source 'C:\Program Files (x86)\Microsoft SDKs\NuGetPackages\'.
  https://api.nuget.org/v3/index.json: Unable to load the service index for source https://api.nuget.org/v3/index.json.
  An error occurred while sending the request.
  The remote name could not be resolved: 'api.nuget.org'
. Please see Error List window for detailed warnings and errors.				

issue 4
Severity	Code	Description	Project	File	Line	Suppression State
Warning	NU1605	Detected package downgrade: System.IO.FileSystem from 4.3.0 to 4.0.1. Reference the package directly from the project to select a different version. 
 Xamarin.UWP -> Xamarin -> MAVLink -> log4net 2.0.8 -> System.Net.Sockets 4.1.0 -> runtime.win.System.Net.Sockets 4.3.0 -> System.IO.FileSystem (>= 4.3.0) 
 Xamarin.UWP -> Xamarin -> MAVLink -> log4net 2.0.8 -> System.IO.FileSystem (>= 4.0.1)	Xamarin.UWP	E:\Ashish2\MissionPlanner\ExtLibs\Xamarin\Xamarin.UWP\Xamarin.UWP.csproj	1

magicrub, 2019-07-22:
Which wiki page did you follow to compile?

ashish-yadav1, 2019-07-22:
https://github.com/ArduPilot/MissionPlanner.git

ashish-yadav1, 2019-07-22:
Severity	Code	Description	Project	File	Line	Suppression State
Error	NU1105	Unable to find project information for 'E:\Ashish2\MissionPlanner\ExtLibs\UsbSerialForAndroid\UsbSerialForAndroid.csproj'. Inside Visual Studio, this may be because the project is unloaded or not part of current solution. Otherwise the project file may be invalid or missing targets required for restore.	MissionPlanner.Stats	E:\Ashish2\MissionPlanner\ExtLibs\MissionPlanner.Stats\MissionPlanner.Stats.csproj	1	

UsbSerialForAndroid is not compatible in visual studio 2017

mathisgag, 2019-08-23:
Same problem...

meee1, 2019-08-24:
install VS with the mobile build tools

mathisgag, 2019-08-30:
Worked. this problem is solved. I'll look in the other issues for my other problems.