ArduPilot/MissionPlanner #2187: Cannot edit Flighdata.cs

kane111, 2019-07-30:
#### Issue details
Cannot edit Flightdata.cs in design view

#### Version
Latest Master

I'm trying to add a widget to the Flightdata to view some data from the distance sensor. However I cannot open the flightdata class in design view. There are a number of errors associated. 

** Could not load file or assembly 'SkiaSharp, Version=1.60.0.0, Culture=neutral,  PublicKeyToken=0738eb9f132ed756' or one of its dependencies. The system cannot  find the file specified | Could not load file or assembly 'SkiaSharp, Version=1.60.0.0, Culture=neutral,  PublicKeyToken=0738eb9f132ed756' or one of its dependencies. The system cannot  find the file specified
-- | --
Could not load file or assembly 'SkiaSharp, Version=1.60.0.0, Culture=neutral,  PublicKeyToken=0738eb9f132ed756' or one of its dependencies. The system cannot  find the file specified **

Also, I am unable to build without unloading the Mock project.

ssdgigabyte, 2019-08-04:
Me too I have this problem because  SkiaSharp needed 1.60.0 but Install 1.68.0
[image]

highfreq, 2019-09-02:
Same problem here, tried to downgrade to 1.60.0 but than i get a ton of different errors.
Did any of you solve the issue?