ArduPilot/MissionPlanner #2188: Motor Test ABCD does not equal 1234

notreallyaclue, 2019-07-31:
#### Issue details
On the motor test screen there is buttons for motor a, motor b, motor c, motor d. There is a web link to check configuration on the wiki but this displays photos with layout 1,2,3,4. The 2 are not interchangeable. 

When used abcd follow clockwise, where motor 1234 follow diagonally. A suggested fix would be either:
Change the buttons to actuate via the motor numbers, 
Align a=1, b=2, c=3, d=4,
(Easiest) Add a note on the motor test page describing the issue.

#### Version
1.3.68 build 1.3.7105.26478

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quad, X

#### Hardware type
F405 CTR

#### Logs
N/A

rmackay9, 2019-07-31:
For better or for worse this is done on purpose.  The idea is that the motor orders are confusing.. so after the setup the user should push the buttons A, B, C, D, etc and check that each motor (checking in a clockwise direction) turns.  Regardless of the frame type, the order is always clockwise when using the Motor Test.

I think I'll close this issue if that's OK.

closed 2019-07-31 by rmackay9

notreallyaclue, 2019-07-31:
Understood, but would be valuable information to have that explained on the motor test page of mission planner.