ArduPilot/MissionPlanner #2191: Signing issue when compiling mission planner

galdalali, 2019-08-06:
#### Issue details
I'm trying to compile mission planner and got the following errors in updater project:
`The command ""C:\Program Files (x86)\Windows Kits\10\App Certification Kit\signtool.exe" sign /n "Michael Oborne" /t http://timestamp.verisign.com/scripts/timestamp.dll "C:\Users\Gal\Documents\bit\MissionPlanner\Updater\bin\Debug\Updater.exe"" exited with code 1.`
and
`No certificates were found that met all the given criteria.
`
The "Sign the ClickOnce Manifests" checkbox is unchecked.
How can I solve this issue?

Thanks, Gal
		

#### Version
1.3.68

meee1, 2019-08-06:
remove the post build command.
I will not be distributing the cert used to sign the updater

GBJin, 2019-08-12:
I remove all the post build command but still got the certificates errors.
[image]
I also got other errors below:
[image]
I installed the latest MP version 1.3.68
[image]
The main code is downloaded from the website, which should be the latest.
Do anyone have the issues? I used to compile MP 1.3.56, but last week I downloaded for the latest code and try to compile, it come up with these errors. Most likely due to the Xamarin project, but I can't solve it.

meee1, 2019-08-12:
remove the post build commands in the updater project

GBJin, 2019-08-12:
Thanks, certificate errors solved.
But I still have errors like below:
[image]
Cannot resolve Assembly or Windows Metadata file 'C:\MissionPlanner-master (1)\MissionPlanner-master\ExtLibs\Xamarin\Xamarin\bin\Release\netstandard2.0\Xamarin.dll'	
I tried to find Xamarin.dll, but there's not such dll file in either installed MP folder or the main code folder download from Github. What did I miss?
I installed .NET Core SDK 2.2.108. And pretty much checked every items which need to be checked.
[image]

meee1, 2019-08-12:
install the mobile components

GBJin, 2019-08-13:
Problem solved, Thanks for the reply!

closed 2019-08-14 by meee1

ojas032, 2020-01-14:
> remove the post build commands in the updater project

I could not find Updater project