ArduPilot/MissionPlanner #2194: Mirror mavlink stream on serial port Broken

macfly1202, 2019-08-10:
#### Issue details
in mission planner, ctrl+F, mirror mavlink stream to output serial doesn't work anymore (to com1, 57600 or 115200, write access ticked or not).
I use this feature of outputting mavlink stream for antenna tracker and externall small converter use and this feature is so nice (yaapu debug/test/improvement and so on)
If it could be be debugged, it will be super nice.  

#### Version
1.3.68.1 build 1.3.7160.16169

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
plane in sitl

#### Hardware type
sitl

#### Logs
no logs

meee1, 2019-08-10:
i just checked this via tcp, and is working fine. so i will check via serial tomorrow.

macfly1202, 2019-08-10:
@meee1 thanks !

macfly1202, 2019-09-08:
Hi, I redo some test today with 2 different hardware. One from zs6buj that is able to decode mavlink stream, and another device, the old Ardustation Mega from JDrones that was working nice previoulsy. These 2 device was previously able to decode Mavlink over serial from the top ctrl+f Mavlink menu. Serial connection is opened but there is no heartbeat it seem or other readable packet. I try different speed in case but none of them work with or without 'write access' enable.

KR