ArduPilot/MissionPlanner #2197: No way to exit a TakeOff command from the context menu

votingprawn, 2019-08-13:
#### Issue details
This is a minor UX issue that can cause a more dramatic issue when coupled with an apparent bug in ArduPlane (https://github.com/ArduPilot/ardupilot/issues/12035)

1) When you select TakeOff from the right-click context menu, the altitude modal that appears is identical to the Guided Alt one, which means you can think you're setting a guided mode altitude when you're actually issuing a TakeOff command.

2) There appears to be no way to stop the TakeOff happening once you select it from the menu, by that I mean pressing "Cancel" or "X" on the altitude modal results in a TakeOff instruction happening anyway - I'm not sure if this is intentional behavior or not.

#### Version
1.3.68.1 build 1.3.7160.16169

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Sub

magicrub, 2019-08-13:
This is likely vtol only. Please confirm and if so please change the issue info.

votingprawn, 2019-08-13:
I've checked in SITL with plane, and the TakeOff context menu also says "Enter Guided Mode Alt" and pressing cancel also results in the command being sent to the vehicle, shifting it from whatever mode it was in to guided mode.