ArduPilot/MissionPlanner #2200: MAVLinkFTP for components

olliw42, 2019-08-15:
I'm not sure if this is a issue or should be a support request, but I suspect former, sorry if I'm mistaken.

I do have a MAVLink component connected to a ArduPilot flight controller which in turn is connected to MissionPlanner, and this component (partially) supports MAVLinkFTP. For instance it sends AUTOPILOT_VERSION when requested with the MAV_PROTOCOL_CAPABILITY_FTP set.

However, whatever I do and wherever in MP I search, I fail to get any MAVLinkFTP option or feature. I'm suspecting that maybe MAVLinkFTP is available only for autopilots and not components, or that some additional conditions need to be meet for it to become enabled, which my component doesn't meet.

Any enlightment in this matter would be much appreciated.

THX.

MissionPlanner latest version: 1.3.68.1 build 1.3.7166.638

meee1, 2019-08-15:
currently mavftp is hidden behind control-x, make sure you select your component as the target first, and it should work.

olliw42, 2019-08-15:
uuupppsss ... I thought to really have tried all control-key combinations which my keyboard allows more than few times ... but I really must have missed this one ... that's a bit embarrassing now ...

MANY thx, also for the super quick response

I close.

closed 2019-08-15 by olliw42