ArduPilot/MissionPlanner #2201: New Flight Logic Command DO_Turn

LOGANUASBVLOS, 2019-08-15:
#### Enhancement details
A desired feature would be a Logic command like DO_Turn_Left and DO_Turn_Right. The inability to determine and command a turn direction is an issue for Beyond Visual Line of Sight Flight Operations within a defined Geo Fence operations area. The unpredictability of turn direction is undesirable and can potentially result in the loss of aircraft in certain situations. 

#### Version
Current version in use is 1.3.63

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Fixed wing

#### Hardware type
Pixhawk

#### Logs
none

magicrub, 2019-08-15:
This is handled already in several different ways, all involve GUIDED mode. Are you familiar with that flight mode?

- MAV_CMD_DO_REPOSITION
- MAVLINK_MSG_ID_SET_ATTITUDE_TARGET
- MAVLINK_MSG_ID_SET_POSITION_TARGET_LOCAL_NED
- MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT
- param STICK_MIXING with RC input or mavlink overrides via joystick input to MP

With Mission Planner you can click on the screen and make the plane go wherever you want.

What are you asking for, Left/Right arrow keys?

LOGANUASBVLOS, 2019-08-15:
I am somewhat familiar with guided mode, however auto is preferred for operational purposes. It is more for in the event of lost link C2 failure, to have a set command in place for the aircraft to make on its own. In a lost link event it will be sequenced to an appropriate planned path, but as far as i can tell there is not option to add a commanded turn logic for a turn direction. 

I am asking for a feature similar to the Do_jump within mission planning that would allow for an automatic commanded logic to the autopilot.

magicrub, 2019-08-15:
If I understand your right, in the event of a lost link you want to send a command over the link to do something?

What you want is FS_SHORT_ACTN and FS_LONG_ACTN.

See
http://ardupilot.org/plane/docs/apms-failsafe-function.html

LOGANUASBVLOS, 2019-08-15:
You have pointed to the feature that initiates the lost link sequencing, but what I am requesting would add further control to the sequencing. For example, Do_land_start --> Do_Turn_Left / right ----> Do_Jump. The Reason a feature like this would significantly help is for Beyond Visual Line of Sight Flight Operations. The operations take place within Specifically defined Geo fences that bend and contort in places. In those contorted places in the event of FS_Long_ACTN a turn in the direction of the contorted Fence boundary will lead to a breach, and will cause the aircraft to override FS_Long_ACTN logic and fly in Guided mode to nearest Rally point. This Will lead to the aircraft being lost link and stuck loitering around a rally point possibly miles away.

magicrub, 2019-08-15:
all accomplished via multiple DO_LAND_STARTs. It has a lat/lng, so if there's more than one it will jump to the one that is closest to the vehicle.

LOGANUASBVLOS, 2019-08-16:
I understand that it will go to the closest one, but it would be helpful if i could have a set command in place to only turn left or only turn right in the sequencing after the closest DO_LAND_START is triggered.