ArduPilot/MissionPlanner #2205: videowriter

ashish-yadav1, 2019-08-19:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2019-09-04:
this is by design. there are too many device that dont have enough cpu to use a better codec. you can reencode as you require however

closed 2019-09-04 by meee1

ashish-yadav1, 2019-09-28:
How can I get Google Earth 3D features in FlightPlanner
and Map Switch From 2D to 3D
like this