ArduPilot/MissionPlanner #2208: DF Log viewer shows mode numbers only

vierfuffzig, 2019-08-24:
Issue: LogViewer does not display ArduPlane flight mode names, only numbers

[image]

Version: latest

Platform: ArduPlane 3.10dev / current master

Hardware type: OmnibusF4ProV3 and MatekF405-Wing

Logs: 
https://github.com/vierfuffzig/Chibios-miscellaneous/raw/master/ASW28_omnibus.BIN
https://github.com/vierfuffzig/Chibios-miscellaneous/raw/master/Talon_MatekF405Wing.BIN

meee1, 2019-09-04:
what MP version?

vierfuffzig, 2019-09-04:
thanks for looking into this!
first noticed on 1.3.68.1 build 1.3.7174.16110  through 1.3.68.1 build 1.3.7186.15082 of today.
i found it depends on fw version the log was acquired with too. respective logs written on plane master of the last few weeks.
[this log](https://github.com/vierfuffzig/Chibios-miscellaneous/raw/master/nanoT.BIN) written with plane 3.9.8 displays mode names correctly.

vierfuffzig, 2020-06-30:
this has been fixed along the way, so closing.

closed 2020-06-30 by vierfuffzig