ArduPilot/MissionPlanner #2211: Mission Planner on Android

Wolfgang-BN, 2019-08-27:
Sorry, this isn't an issue,

but Mission Planner is the best of all the ground station controll software!!
Is there any chance, that MP will run on Android in the future? I think, i'm not the only one with this burning desire.

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
All
#### Hardware type
All

kikislater, 2019-08-29:
C# and .net : no way that it could be on Android ... 
Use Qgroundcontrol

meee1, 2019-08-29:
not true, ive been working on something in the background. still a while away though

kikislater, 2019-08-29:
@meee1 it could not be easily port. Glad to hear but will be totally different in term of architecture. Good luck !

kikislater, 2019-09-01:
So you will use Xamarin ? First project I heard that will use this ... Looking forward how it will be !

meee1, 2019-09-02:
[image]

MR444, 2021-01-01:
Happy new year to everyone!

I just found and tested the missionplanner App on Android.
https://play.google.com/store/apps/details?id=com.michaeloborne.MissionPlanner&hl=en
It looks like he put the MP on Android . Not really easy to use without mouse. I tried on 10 inch Lenovo tablet, so I think almost impossible to use on phones.

Connect/Disconnect Button does not work (only once). Crashes sometimes when I want to use missonmode but proably will be improved in future.
Discussions: https://discuss.ardupilot.org/t/mission-planner-android/62896/48