ArduPilot/MissionPlanner #2213: SDK not compatible with Core 2.2 and other issues with 1.3.68 on VS 2017

mathisgag, 2019-08-30:
#### Issue details
I downloaded the latest release with git. (Actually I downloaded the commit 34c040dd witch is the commit for the 1.3.68 version.) I opened it with VS 2017, with the mobile dev tools. And I got first off an error telling me that "The current .NET SDK does not support targeting .NET Core 2.2.  Either target .NET Core 2.0 or lower, or use a version of the .NET SDK that supports .NET Core 2.2." with Mock. It's odd because I don't think there's a 2.2 version of the .NET Framework. I also got an other error witch I don't understand telling something about the Package Microsoft.AspNetCore.Blazor.Cli 0.5.1 . 
I then had errors for the OpenTK missing.
Also other reference missing (GraphicsMode, IWindowInfo, GraphicsContext)
An ambigous call.
And finally strangely some missing files witch are not in the master nor in the release. They are called Xamarin.dll. I think it also cause an error with the ResolveLibraryProjectImports.
I linked the complete error log in a .txt file.

#### Version
1.3.68

#### Platform
[  ] All
[ x ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
hexa

#### Hardware type
pixhawk

#### Logs
[ErrorLog_MissionPlanner_1.3.68.txt](https://github.com/ArduPilot/MissionPlanner/files/3561303/ErrorLog_MissionPlanner_1.3.68.txt)
I hope you can explain to me what happened.
Thx

meee1, 2019-08-31:
install dotnet core 2.2

mathisgag, 2019-08-31:
I had also installed VS 2019, witch caused the dotNET 3.0 to be installed, so I had to uninstalled it in order to target 2.2.
Thanks! 
It got rid of some errors:
[ErrorLog_MissionPlanner_1.3.68_2.txt](https://github.com/ArduPilot/MissionPlanner/files/3562567/ErrorLog_MissionPlanner_1.3.68_2.txt)

mathisgag, 2019-08-31:
So it appears that I have to errors:
1. Missing Xamarin.ddl file
2. Unknow reference to the assembly OpenTK in the Xamarin project:
[image]
It dosen't have a path like others projects.
Probably that OpenTK is the only error left. And that the Xamarin.dll file couldn't be created because the project Xamarin had errors.
Some errors appeard after a while:
[ErrorLog_MissionPlanner_1.3.68_2.txt](https://github.com/ArduPilot/MissionPlanner/files/3562605/ErrorLog_MissionPlanner_1.3.68_2.txt)

willhuo, 2019-09-03:
I come across the same problem that the solution can not compile with not found opentk.
This package is located in ExtLibs folder,just a dll file.

meee1, 2019-09-03:
you dont need this project to compile mission planner.
compile just mission planner

mathisgag, 2019-09-03:
Ok, so I don't do a batch build on the entire solution but a build/rebuild on the project mission planner.
It works just fine! thanks!

mathisgag, 2019-09-03:
I will close this issue.

closed 2019-09-03 by mathisgag