ArduPilot/MissionPlanner #2219: SITL simulator reports pre-arm warning: throttle below failsafe

gzhang8, 2019-09-14:
#### Issue details
When launching simulator from MP under quad mode, we see this pre-arm warning and not able to do further flight simulation. 

#### Version
1.3.66
#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Simulated quad
#### Hardware type
SITL
#### Logs
[image]

dhollenbeck1, 2019-09-14:
The plane does not work either.

dhollenbeck1, 2019-09-14:
Found a commit in the AP_HAL_SITL layer: 
https://github.com/ArduPilot/ardupilot/commit/1735563bb790ff70b346b1bf5c5342afb34f61c2

Seems to be the issue we are experiencing here.

rmackay9, 2019-09-16:
Sorry for the troubles, a PR got through without enough testing and review.  It's been reverted in master now so I think it's OK to close this issue.

closed 2019-09-16 by rmackay9