ArduPilot/MissionPlanner #2221: Correct behavior on received PARAM_VALUE/PARAM_EXT_VALUE/PARAM_EXT_ACK messages

olliw42, 2019-09-18:
according to MAVLink spec, the parameters should also be updated whenever a  PARAM_VALUE/PARAM_EXT_VALUE/PARAM_EXT_ACK message is on the link indicating that some other component than a GCS/MP has changed a parameter. The documents have been or are in the process of being changed to make that clear.

QGC does this, MP currently doesn't do this

it was suggested that I bring this to your attention so you could adopt the correct behavior (which I herewith did LOL)

background issues/PRs
* Pl clarify behavior on received PARAM_VALUE/PARAM_EXT_VALUE/PARAM_EXT_ACK messages #1181
* Parameters: Clarify how parameter lists maintained #204
* Add docs for extended parameter protocol #210

meee1, 2019-09-18:
MP already does this for PARAM_VALUE, not the other states though

olliw42, 2019-09-18:
thx for the clarification
I indeed had tested this only for the PARAM_EXT messages as part of the camera microservice
I changed the title

meee1, 2019-09-18:
looks like PARAM_EXT does not exist in ardupilot at all yet

olliw42, 2019-09-18:
that's correct AFAIK
not only this is missing, but many newer MAVLink additions are also not in AP's common.xml
but that's a story by its own,