ArduPilot/MissionPlanner #2223: MissionPlanner reports params written successfully while failed

Jaaaky, 2019-09-26:
#### Issue details
MissionPlanner reports that params has been written successfully while it actually failed.
This occurs in 2 cases:
1. When a connection problem occurs, you change a param then "write" it says at 1st param "blabla" failed. And then says "Parameters has been written successfully"
2. When writing an Internal or Read-only param like "MIS_TOTAL", it doesn't send any error at all, but says "Parameters has been written successfully". Do a "refresh params", you'd find that param has not been written.

#### Version
_1.3.68.1_

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2019-10-05:
i will fix point 1
point 2 i will leave as is. as its readonly, and should not be changed anyway.

Jaaaky, 2019-10-05:
Point 2 is not random().. Writing a to a read_only or internal_use_only param returns an error, but MP doesn't catch it.
mavproxy shows the error regularly. MP does not.