ArduPilot/MissionPlanner #2228: Don't use source sys & comp ID = 0 in MAVLink header

olliw42, 2019-10-07:
#### Issue details
for a QUADROTOR a "Trigger Now" makes MP sending out a MAVLink command MAV_CMD_DO_DIGICAM_CONTROL with source system ID = 0 and source component ID = 0 in the header.

This is not in accord with the MAVLink spec (the spec says so already for the system ID, and is in the course of being updated to say so also for the comp ID).

Note: The target sysID and compID of the command message are also set to 0, but this is OK for broadcasting.

Note II: For a GIMBAL or a CAMERA, MP sends out the same message, but now with the message's source and comp IDs correctly set to MP's system ID and component ID (normally 1,190) (and the target system set to the sys&comp ID of the GIMBAL or CAMERA, i.e., non-broadcasting)

#### Version
1.3.68.1 build 1.3.7166.638

meee1, 2019-10-07:
MP should send the message to whatever is the current active node. im not sure how it could send from a 0,0 ID, as this is pretty much hardcoded. do you have a tlog with the example?

olliw42, 2019-10-07:
this happens only when the current active node is QUADROTOR (i.e. autopilot). I suspect it's related to that the target sysID&compID are set to 0,0 for that case.

I'll try to produce a tlog

olliw42, 2019-10-07:
I hopefully got it right
www.olliw.eu/drop/michael-mp/2019-10-07_10-18-40.tlog
should show two components, a quadrotor and a camera, and should show "trigger now" for first the quadrotor and then for the camera as active node

meee1, 2019-10-14:
ok so i had a look at the log, and i dont see any issue
is see sysid 2, compid 1 and 100

and this is the do command longs

2019-10-07T16:19:13.712 | FD | 20 | 0 | 0 | 53 | FF | BE | 4C | mavlink_command_long_t | param1 | 0 | param2 | 0 | param3 | 0 | param4 | 0 | param5 | 1 | param6 | 0 | param7 | 0 | command | 203 | target_system | 2 | target_component | 1 | confirmation | 0 |   | sig | Len | 44 | crc16 | 32517
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2019-10-07T16:19:13.727 | FD | 4 | 0 | 0 | E6 | 2 | 1 | 4D | mavlink_command_ack_t | command | 203 | result | 0 |   | sig | Len | 16 | crc16 | 43252 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
2019-10-07T16:19:13.836 | FD | 4 | 0 | 0 | F7 | 2 | 64 | 4D | mavlink_command_ack_t | command | 203 | result | 0 |   | sig | Len | 16 | crc16 | 61282 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
2019-10-07T16:19:18.657 | FD | 20 | 0 | 0 | 59 | FF | BE | 4C | mavlink_command_long_t | param1 | 0 | param2 | 0 | param3 | 0 | param4 | 0 | param5 | 1 | param6 | 0 | param7 | 0 | command | 203 | target_system | 2 | target_component | 100 | confirmation | 0 |   | sig | Len | 44 | crc16 | 46694
2019-10-07T16:19:18.672 | FD | A | 0 | 0 | FD | 2 | 64 | 4D | mavlink_command_ack_t | command | 203 | result | 0 |   | sig | Len | 22 | crc16 | 61952 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |

olliw42, 2019-10-14:
man thx

I am highly confused

I have to say that I did not test the output of MP directly, but the output send from the autopilot to my component. I assumed that it would forward whatever MP sends out, anything else was out of my mind. The data going to my component I have checked in two ways, by adding respective debug code to my component and by directly sniffing the uart data line, and both showed the same which made me quite confident.

Something is strange here. I will investigate this further, and especially also will look now directly at MPs output.

Thanks for your time and sorry if I should have wasted it.

meee1, 2019-10-14:
it is possible the autopilot is doing it, as the target would be the autopilot, not the camera so anything forwarded by the autopilot would have the wrong id in it anyway.

it depends on what you expect the default to be

olliw42, 2019-10-14:
I'm not sure though the autopilot should do that. I understand it might replace the source by it's own, but think it should not set the source to zeros. 
As said, I will investigate this further.

Just as a side note: The MAVLink spec indeed was somewhat inaccurate as regards the source IDs, and it was corrected only very recently (now zeros are explicitly not allowed in the source). QGC also had this issue and corrected it very recently. So, you might well be right with the autopilot (and I bite my a.. that I haven't looked at MP's output directly). The expected default however should now be claryfied. :)

thx again

olliw42, 2019-10-16:
I can confirm that it's the autopilot (ArduPilot) and not MP. MP is perfectly fine.

sorry for not having considered and tested this possibility before. And many thx.

I accordingly close this.

closed 2019-10-16 by olliw42