ArduPilot/MissionPlanner #2241: Geofence Breach displayed when no actual breach?

rmackay9, 2019-10-31:
I think the "Geofence Breach" message may not be being cleared correctly.  This is testing with Copter-4.0.0-dev and latest beta MP.

Below screen shots show what I mean.  The "Geofence Breach" is displayed but as far as I can tell the "FENCE_STATUS" message being sent is showing there are no actual breaches.

[image]
[image]

meee1, 2019-10-31:
the message comes from 
MAV_SYS_STATUS_SENSOR

so i asume there is a reporting bit error in there

rmackay9, 2019-10-31:
@meee1, ok, I see the issue.  It's all these extra checks that have been added. https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Fence/AC_Fence.cpp#L455

closed 2019-10-31 by rmackay9