ArduPilot/MissionPlanner #2242: GStreamer initialization failed

linkevinlin1, 2019-10-31:
#### Issue details
_Please describe the problem, or desired feature_
After confirming the GStreamer source, it starts to download GStreamer and give this error message.
[image]

If I install GStreamer (tried 1.16.1 & 1.9.90) beforehand, this error pops out.
[image]

Please help me solve this issue.
Thanks.

#### Version
_What version was the issue encountered with_
MP 1.3.68.1 build 1.3.7421.30827
#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2019-10-31:
You appear to be using a 32bit os. The version shipped with mission planner is 64 bit only

linkevinlin1, 2019-11-01:
Thanks for the reply.

I'm using a 64bit os. I also have this issue on two other 64bit computers. Could it be not compatible to some libraries on our computers?