ArduPilot/MissionPlanner #2243: MissionPlanner VS2017,2019, Win7 Win10 Compilation

fengzhu2000, 2019-11-01:
#### Issue details
NU1701	Package 'GDAL 1.11.1' was restored using '.NETFramework,Version=v4.6.1' instead of the project target framework '.NETStandard,Version=v2.0'. This package may not be fully compatible with your project

sln:
--ExtLibs:
  --MissionPlanner.Utilities:
    --Dependencies
       --Nuget:
            --GDAL(1.11.1):
                --Package 'GDAL 1.11.1' was restored using '.NETFramework,Version=v4.6.1' instead of the project target framework '.NETStandard,Version=v2.0'. This package may not be fully compatible with your project

#### Version
_MissionPlanner 1.3.68 Source code_

meee1, 2019-11-01:
This is a warning not an error. Or update your visual studio version

fengzhu2000, 2019-11-05:
The problem I made is that I deleted the folder (newget).  after recovered the original folder,most problems disappeared.   use VS2019 or 2017 comes the same problem.  Thanks for your help, at least MP can passed compiled . Thank you.