ArduPilot/MissionPlanner #2244: MissionPlanner  MSB3577 Two output file namesresolved to the same output path

fengzhu2000, 2019-11-01:
#### Issue details
 --MSB3577	Two output file names resolved to the same output path: "obj\Debug\netstandard2.0\MissionPlanner.Strings.resources"	
 --Project: MissionPlanner.Strings  _
 --File: C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets	
 --Line 3049

--MSB3577	Two output file names resolved to the same output path: "obj\Debug\net461\MissionPlanner.ConnectionOptions.resources"	
 --Project:     MissionPlanner	
 --File:     C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets	
 --Line:    3049
#### Version
_MissionPlanner SourceCode 1.3.68_

meee1, 2019-11-01:
Use windows 10 to compile will fix this error

fengzhu2000, 2019-11-05:
Yes,really so, thank you. But I am still doubting why there is such problem.Thank you.

davidbitton, 2020-05-10:
I'm having the same problem as @fengzhu2000. @meee1 are saying that I can build on Windows and use the EXE on mac with the `dotnet` command line command?

davidbitton, 2020-05-11:
@meee1 what's the story with the XamarinUI project? Have you hack at this being a Core app and not a Framework app? Thanks.

meee1, 2020-05-11:
Yes. The exe on all platforms is the same. Exe from windows run on mono for Mac and linux

davidbitton, 2020-05-11:
Hmm. Just to remove an uncertainty, I downloaded the latest zip, and launched the exe, but not so good on the response. I'm on a Macbook Pro running Catalina (10.15.4). As an aside, I was able to build the code in my Win10 VM. Is there anything you'd like me to try? Thanks.
### mono -V:
```
➜  MissionPlanner-latest mono -V
Mono JIT compiler version 6.8.0.123 (2019-10/1d0d939dc30 Thu Mar 12 23:19:08 EDT 2020)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:
	SIGSEGV:       altstack
	Notification:  kqueue
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug
	Interpreter:   yes
	LLVM:          yes(610)
	Suspend:       hybrid
	GC:            sgen (concurrent by default)
```

### program output:
```
➜  MissionPlanner-latest mono MissionPlanner.exe
If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
To fix any filename case issues under mono use    export MONO_IOMAP=drive:case
Data Dir /Users/davidbbitton/Mission Planner/
Log Dir /Users/davidbbitton/Mission Planner/logs
Running Dir /Users/davidbbitton/Downloads/MissionPlanner-latest/
User Data Dir /Users/davidbbitton/Mission Planner/
WARNING: The Carbon driver has not been ported to 64bits, and very few parts of Windows.Forms will work properly, or at all
INFO Program - ******************* Logging Configured *******************
ERROR Program - System.EntryPointNotFoundException: LoadLibrary assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) MissionPlanner.Utilities.NativeLibrary.LoadLibrary(string)
  at MissionPlanner.Program.Start (System.String[] args) [0x00258] in <a50d9501ac1e4291b857e1e537bbfe30>:0

=================================================================
	Native Crash Reporting
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries
used by your application.
=================================================================

=================================================================
	Native stacktrace:
=================================================================
	0x1014dc1cb - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_dump_native_crash_info
	0x1014747a5 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_handle_native_crash
	0x1014d62fe - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : altstack_handle_and_restore
	0x7fff67fad25e - /System/Library/PrivateFrameworks/SkyLight.framework/Versions/A/SkyLight : SLDisplayBounds
	0x109e6304d - Unknown
	0x106605856 - Unknown
	0x1013d167b - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_jit_runtime_invoke
	0x1015e0552 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_runtime_try_invoke
	0x1015df178 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_runtime_class_init_full
	0x1013c9190 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_jit_compile_method_inner
	0x1013ccb7a - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_jit_compile_method_with_opt
	0x101477387 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : common_call_trampoline
	0x101476dc0 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_magic_trampoline
	0x101895393 - Unknown
	0x109c918eb - Unknown
	0x109a3201f - Unknown
	0x109a31c67 - Unknown
	0x1019750b3 - Unknown
	0x1013d167b - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_jit_runtime_invoke
	0x1015deb67 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_runtime_invoke_checked
	0x1015e71ee - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : do_exec_main_checked
	0x101432297 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_jit_exec
	0x1014354ad - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : mono_main
	0x1013c05c8 - /Library/Frameworks/Mono.framework/Versions/Current/Commands/mono : main
[... 25 lines cut ...]
	Managed Stacktrace:
=================================================================
	  at <unknown> <0xffffffff>
	  at System.Windows.Forms.XplatUICarbon:CGDisplayBounds <0x000bc>
	  at System.Windows.Forms.XplatUICarbon:get_WorkingArea <0x00072>
	  at System.Windows.Forms.XplatUICarbon:get_VirtualScreen <0x00043>
	  at System.Windows.Forms.XplatUI:get_VirtualScreen <0x00048>
	  at System.Windows.Forms.Screen:.cctor <0x0017a>
	  at System.Object:runtime_invoke_void <0x000a5>
	  at <unknown> <0xffffffff>
	  at System.Windows.Forms.Form:get_CreateParams <0x0062a>
	  at System.Windows.Forms.Control:CreateHandle <0x0009e>
	  at System.Windows.Forms.Form:CreateHandle <0x00052>
	  at System.Windows.Forms.Control:CreateControl <0x000c6>
	  at System.Windows.Forms.Control:SetVisibleCore <0x00112>
	  at System.Windows.Forms.Form:SetVisibleCore <0x00252>
	  at System.Windows.Forms.Control:set_Visible <0x00052>
	  at System.Windows.Forms.Control:Show <0x00042>
	  at System.Windows.Forms.Control:Show <0x000a2>
	  at MissionPlanner.Program:Start <0x0103a>
	  at MissionPlanner.Program:Main <0x00042>
	  at <Module>:runtime_invoke_void_object <0x000b0>
=================================================================
[1]    14134 abort      mono MissionPlanner.exe
```

meee1, 2020-05-11:
you need to try mono32 on mac. but i dont think it works

davidbitton, 2020-05-12:
Yeah, that didn't work:
```
➜  MissionPlanner-latest mono32  MissionPlanner.exe
zsh: bad CPU type in executable: mono32
```
and using `mono64` generated the same error(s) as before. Thanks.

davidbitton, 2020-05-12:
@meee1 I launched MP with mono in debug:

    MONO_LOG_LEVEL=debug mono MissionPlanner.exe 2>&1 1> log.txt

I won't flood you with the output, but the errors about User32 missing were enough to find this:

[Getting System.DllNotFoundException for user32.dll in VS 2019 Mac](https://tro.pe/3625Jx7)

Not too sure what to do at this point as it would require refactoring. Well, at least I found some insight.

meee1, 2020-05-12:
MP use to run on older versions of OSX, but needed XQuartz.

and also
WARNING: The Carbon driver has not been ported to 64bits, and very few parts of Windows.Forms will work properly, or at all

it will currently never work.