ArduPilot/MissionPlanner #2245: Gps in mission planner/osd wrong, correct in radio

tj51, 2019-11-02:
#### Issue details
I have gps date coming out of pixhawk  to my x12s vi telem2 show correct verified with cell data,
mission planner and osd data incorrect.

#### Version
_What version was the issue encountered with_
3.57
#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk radiolink mini pix
#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2019-11-04:
need more info.
a log
the difference in time shown? etc

tj51, 2019-11-04:
Hay Mike,?? All I can think if is to take pictures of my radio and the 
osd screen, will this help you understand?, Tj