ArduPilot/MissionPlanner #2246: Connects then disconnects on launch

jimovonz, 2019-11-03:
On launch, MP automatically attempts to connect using the latest connection parameters. If a controller is attached, then it starts downloading parameters only to disconnect half way through. This then requires that the parameters be downloaded again once a connection is manually initiated. This adds an annoying delay to the connection process. This behaviour has been present for as many versions of MP as I can remember and is still present on the current beta version.  Ideally, automatic connection on launch would be a configurable parameter and when enabled, not drop out when almost complete.

meee1, 2019-11-04:
What protocol are you using? UDP?

jimovonz, 2019-11-04:
Yes UDP