ArduPilot/MissionPlanner #2247: Copter-4.0 updates to extended tuning screen

rmackay9, 2019-11-04:
As part of the Copter-4.0 beta testing, we've noticed that a few fields on the Extended tuning page are blanked out because we've changed parameter names.

[image]

The TUNE_ parameters should be TUNE_MIN and TUNE_MAX (was TUNE_LOW, TUNE_HIGH)

For the Filters the single parameter for each axis has been replaced with 3 filters.  So _FILT -> FLTE, FLTD and/or FLTT.  After a discussion with @lthall we think it would be best if all three were shown.

@lthall is going to come up with some more input for this screen.

lthall, 2019-11-04:
[image]

I would suggest adding:
ATC_ACCEL_P_MAX
ATC_ACCEL_R_MAX
ATC_ACCEL_Y_MAX
ATC_INPUT_TC

rmackay9, 2019-12-27:
@meee1, we're probably 1 week away from the Copter-4.0.0 release so it would be good to get these sorted out.  I think the Tune change in particular should be pretty straight forward because we've just renamed TUNE_HIGH to TUNE_MAX and TUNE_LOW to TUNE_MIN.

ddomit, 2020-01-23:
This is still not working!