ArduPilot/MissionPlanner #2251: Mission Planner Error

ashish-yadav1, 2019-11-14:
**Mission Planner get Killed** 
The GCSView like FLightdata get BlackOut all the control get Black one by one 
and popup show send error but i can't see the error 

after That Mission Planner get whole Black in and i can't do Anything !!!

rmackay9, 2019-11-18:
I think the 3rd line above is a slight typo, it probably should be, "The popup asks if I want to send the error but then it failed to send".  I've seen this as well.  It's a bit annoying when the user is given a false choice.  I.e. it gives them a choice to send or not but in practice, the sending fails anyway so it was a waste of time to ask and get the user's hopes up.

ashish-yadav1, 2019-11-18:
Any Solution so that Flight data do not get black on all the control and not  get hanged when i fly drone

ashish-yadav1, 2020-01-04:
Hello  Sir,
i just  need some help or guidance currently in mission planner the map is
use  is GMap.Net and its in top view or can i say it is in 2D and also
there is good feature on a click of ctrl +F is 3d MAP but is a drone view
3d  map and i cannot  tilt the map .
so i want help in the map section so i can tilt the map and i can also see
the birds eye view of GMap.Net
i have given a screen short below please help me out

And please guide me please

Thanks