ArduPilot/MissionPlanner #2260: Battery percentage by voltage

jasin755, 2019-11-25:
#### Issue details
Mission planner calculate remaining battery percentage by used mAh and capacity. It is the worst solution ever. Why?

1. When you connect discharged battery you will see 100% == CRASH/FAILSAFE
2. When you modify params and reboot system you will see 100% == CRASH/FAILSAFE
3. When you have many batteries with different capacity and you will forget change capacity == CRASH/FAILSAFE
4. Capacity of old batteries can be smaller == CRASH/FAILSAFE
5. When you have something like 12S10P and you loss 1S or 1P the voltage or capacity will be lower == CRASH/FAILSAFE

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Octocopter

#### Hardware type
Cube black

meee1, 2019-11-26:
this is not something mission planner can do anything about. its a property of the autopilot firmware. MP just displays what its told.
the origin of the information is from mavlink_sys_status_t or mavlink_battery_status_t

jasin755, 2019-11-26:
Ok close it. I will recreate this issue in Ardupilot.

jasin755, 2019-11-26:
Issue moved here: https://github.com/ArduPilot/ardupilot/issues/12897

closed 2019-11-26 by jasin755